uStepper/uStepperS #2: Update readme

ghulands, 2019-02-09:
The readme says to follow the original documentation which talks about the ftdichip vcp. uStepper S uses the Silicon Labs VCP. It might save someone else 10 minutes why the port doesn't show up in the Arduino environment. ;-)

uStepper, 2019-02-10:
Thank you for noticing !
I will have the readme updated soon. are you using a mac?
Best Regards,
Thomas Olsen

ghulands, 2019-02-18:
Yes, using macOS.

sfandersen, 2019-02-19:
I am using Windows 10. It the uStepperS did not show up as a serial port before I downloaded this driver: https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

uStepper, 2019-02-27:
Thanks for pointing out that this issue can also occur on windows 10 !

closed 2019-02-27 by uStepper