uStepper/uStepperS #5: No hard stop

matt-phair, 2019-02-27:
I'm working on implementing my own limit switches, andI might be missing something, but I can't reliably achieve a full hard stop.

`uStepper::stop()` still does a ramp down. At the moment, I'm trying to use something like this:

```
void stop() {
  stepper.driver.reset();
  delay(1000);
  stepper.setup(NORMAL, STEPS_PER_REVOLUTION);
  stepper.setMaxAcceleration(MAX_ACCELERATION);
  stepper.setMaxVelocity(MAX_VELOCITY);
  delay(1000);
}
```

But if this happens in the middle of a very long move, when `stepper.setup(...)` is called, it tries to finish the move it was originally in. If I can reset the current ramp profile, then maybe that will help the reset process?

Or if I'm way off base and there's a simple single line function to call to stop, then can someone point me in that direction?

uStepper, 2019-02-27:
Hi Matt,

A simple solution is to just set the max acceleration to a high value, call the stop function, and set the acceleration value back to normal after the motor has stopped:

`
void stop() {
  stepper.setMaxAcceleration(65535);
  stepper.stop();
  delay(100);
  stepper.setMaxAcceleration(MAX_ACCELERATION);
}
`
I hope this can help solving your issue

matt-phair, 2019-02-27:
@uStepper that solved it, thanks!

closed 2019-02-27 by matt-phair