uStepper/uStepperS #6: Arduino error: Error compiling for board uStepper

derekcovill, 2019-04-30:
Hi there , I recently purchased a ustepper S, and I’ve installed the libraries for this in my Arduino (v 1.8.9...I'm not using the app) and have selected the uStepper board at the bottom of the list of boards, but when I try to run one of the example sketches (e.g. bounce) there is always the following error……….

In file included from
 ...\Arduino\libraries\uStepper_S\Examples\Bounce\Bounce.ino:1:0:

...\Arduino\libraries\uStepper_S\src/uStepperS.h:38:3: error: #error !!This library only supports the ATmega328PB MCU!!

  #error !!This library only supports the ATmega328PB MCU!!

   ^

exit status 1
Error compiling for board uStepper.

Just wondering if you have any advice for this? I’ve tried all the examples from uStepper, UstepperS and UstepperS lite also…

uStepper, 2019-04-30:
Hi Derek,

The problem is that you selected the "ustepper" board. You need to select the "ustepper S" board , if this is the type you have

Best regards
Thomas olsen

derekcovill, 2019-04-30:
Hi Thomas, but there doesn’t seem to be a ustepper S option? [image1.jpeg]

This email has been scanned by MessageLabs' Email Security System
on behalf of the University of Brighton. For more information see:
https://staff.brighton.ac.uk/is/computing/Pages/Email/spam.aspx

This email has been scanned by MessageLabs' Email Security System
on behalf of the University of Brighton. For more information see:
https://staff.brighton.ac.uk/is/computing/Pages/Email/spam.aspx

derekcovill, 2019-04-30:
> Then it is because you only installed the library, and not the hardware package. Please read the "hardware" part under "installation" on the readme page:
> https://github.com/uStepper/uStepperS/blob/master/README.md

I did that but will try it again, perhaps I didn’t do it correctly, fingers crossed and many thanks for your help!

https://staff.brighton.ac.uk/is/computing/Pages/Email/spam.aspx

https://staff.brighton.ac.uk/is/computing/Pages/Email/spam.aspx

derekcovill, 2019-04-30:
Ok, I uninstalled the old one, updated it to the new one and that seemed to work, many thanks!

https://github.com/uStepper/uStepperS/blob/master/README.md

https://staff.brighton.ac.uk/is/computing/Pages/Email/spam.aspx

https://staff.brighton.ac.uk/is/computing/Pages/Email/spam.aspx

closed 2019-05-15 by uStepper