uStepper/uStepperS #7: Changing Hold Current bug

justinchan91, 2019-05-15:
When changing the max current to 80% using:

void setup(void)
{
 stepper.setup(PID);
  delay(1000);
  stepper.setMaxAcceleration(500);
  stepper.setMaxVelocity(800);
  **stepper.setCurrent(80.0);**
}

void loop(void){
  if(!stepper.getMotorState())
  {
    delay(500);
    stepper.moveToAngle(angle);
    angle = -angle;
  }

Any value around 70%-100% gives a random jerking. At start, while moving and sometimes at end where the motor stalls and then continues hap-hazardly.

max value according to datasheet is 31. The driver scales the input float and converts to a range of ints 0-31.

Using a stepper motor rated for up to 3A. Also tried w a smaller stepper motor-same issue.

uStepper, 2019-05-15:
Hi Justin,

Thank you for pointing this issue out. I will look into the cause and come up with a solution.

Best Regards,
Thomas Olsen

greyltc, 2020-02-02:
I see this too.

This isn't just for holding current though, it's for running current. My motors become increasingly unstable and unusable with current values beyond 50%. @uStepper, have you made any progress with solving this bug since last May?

Being limited to 50% of the design & advertised current or less seems like a pretty big deal.

uStepper, 2020-02-04:
Hi,

We have run a long term test on the NEMA 17 motors we sell and use and we can't get the behavior you describe. 

And yes it is reasonable to being limited to 50% max current without adding cooling. There's a lot of heat dissipated from the TMC5130 and thus things will heat up unless you add active cooling. It is the same as for any other stepper driver like the popular StepStick style drivers.

What motors are you using? And what supply voltage?

sriram198198, 2020-02-15:
Hi team,

As you have mentioned that you have tested and was unable to reproduce the behavior. I would like to know what is the max current you have tested with ? 

I am planning to buy few tens of pieces but for my application 50 Percent ( 1 Amp of current is not sufficient in some cases)

I am curious to know If I Could go a bit higher like 60-70 Percent Current so that I get 1.5 Amps to 1.68 Amps out of Ustepper S version.

Thank you,
Sriram.

uStepper, 2020-02-17:
Hi,

So we did an additional test on a motor with a rated current of 2.1A. The test was done using the run continous example. A 24 V lab supply capable of providing 3A was used and for the tests with FAN I used a 24V fan mounted in a fancover. Heatsinks were mounted on all tests.
It is not surprising that the driver gets hot since the on resistance of the TMC5130 can be as high as ~1ohm per half bridge meaning that at full current 4W is developed in heat - which is quite a bit.

Motor: 17HS24-2104S
Room temperature: 22 deg C

Test 1:
Current setting Run/Hold: 50/50 %
Surface chip temp after 30 min: 75.3degree

Test 2:
The same as Test 1 but with fan:
Surface chip temp after 30 min: 42.6deg C

Test 3:
~30min - stopped due to fear of burning the motor
Current setting Run/Hold: 100/100 %
Surface chip temp after ~30 min: 91.3deg C
Motor temperature: 57deg C

None of the test showed any issues with movement

We do not recommend running your driver at high tempepratures. We do recommend that some thought is given on how you use the driver, e.g. lower hold current, do not use brake if not necessary (to avoid heating during standstill) and limit running current to what is necessary. Furthermore, keep the supply voltage as low as you can since a higher supply will generate more heat on the board in general - both on switch mode regulator and driver. So, if it is not strictly needed to run at e.g. 36V then lower your supply voltage to what is actually required. A good starting point is 12V which is sufficient for most applications.

wffbos, 2020-02-26:
Hi, 
I can't find in the documentation how to translate the current which is set in percent to actual current (amperes). I can find this:
http://ustepper.com/docs/usteppers/html/uStepperS_8cpp_source.html#l00392
`void uStepperS::setCurrent( double current )
 {
     if( current <= 100.0 && current >= 0.0){
         // The current needs to be in the range of 0-31
         this->driver.current = ceil(0.31 * current); 
     }else{
         // If value is out of range, set default
         this->driver.current = 16; 
     }
 
     driver.updateCurrent();
 }`
What unit is the value 0-31? Amperes? mili-amperes?
I guess the driver can't be that smart and know the max current for my motor?

uStepper, 2020-02-26:
Hi,

the documentation is here:
[http://ustepper.com/docs/usteppers/html/classuStepperS.html#ac74e343f7ad81aa5c54a68f2514807fe](url)

the input to the function is in percent (0-100), and the driver is a 2A driver, so 0% = 0A and 100% = 2A. 
the 0-31 values you refer to are the input for the register in the driver

no, the driver does not know your motor rating, so you will need to adjust this accordingly

Best Regards,
Thomas Olsen

closed 2020-07-11 by uStepper