uStepper/uStepperS #14: uStepperS::setup()'s "setHome" parameter appears to do nothing

JRVeale, 2019-11-18:
Looking in uStepperS.cpp, I can see `encoder.setHome()` is called twice during `setup()` (on first glance it seems to be being used to determine which direction to set the driver's direction to), regardless of the value of the `bool` parameter `setHome`. It appears that the `setHome` is actually not used anywhere at all.

One may want to use the encoder to receive absolute angles, and therefore wish to prevent encoder's `setHome()` function from being called by setting the value of `setHome` in `setup()` accordingly.

If I've understood the library correctly from my brief look, I suggest changing `setup()` to call `encoder.getAngle()` twice to determine the direction to set for the driver (instead of calling `encoder.setHome()` at the start of `setup()` and then using `encoder.getAngleMoved()`). The final call to `encoder.setHome()` will then only occur if `setup()` has been called with the `setHome` parameter as `true`. This should preserve current behaviour while adding the option to leave the encoder as absolute.

(As an aside, thanks for open-sourcing and I have to say I appreciate how easy it is to skim through the uStepper code and understand the structure of what appears to be going on.)

JRVeale, 2019-11-18:
Actually, I'll try this quickly and if it seems to work without obvious issue I'll open a PR

uStepper, 2019-11-19:
Hi James,

Your suggestion is exactly how it is supposed to work. I don't know how we overlooked this, but the feature is not implemented for some reason. I will add this to the next release !

It is good to hear you are happy with the open-sourcing and the code structure, although i personally think it has gotten pretty messy in most places and could use a clean up ! ;)

Best Regards,
Thomas Olsen

JRVeale, 2019-11-19:
Far be it from me to judge anyone else's code smell!

uStepper, 2020-07-11:
Fixed :)

Best regards,
Mogens

closed 2020-07-11 by uStepper