uStepper/uStepperS #17: SetHome - is my requirement possible

thehelpless, 2019-11-27:
Hi,

I am new to the ustepper and ustepper library.

I am building a camera slider on a linear belt driven rail using a nema 23 and a ustepperS

I want to create a homing routing that does the following (can be on each reset - no need to store long term)

On power on;

move the gantry CW and detect "home" position and set this as "0" steps
move the gantry CCW and detect "end" position and set this as "nSteps" from home.

I want this to be in steps rather than degrees (although I could divide degrees by step angle if needed)

Is the above possible with the ustepper library?

Thanks

uStepper, 2019-11-30:
Hi,
Yes, the above is possible by using the "moveToEnd()" function of the library. This function will move in your desired direction until it hits the endpoint, and return the distance from your starting point to the endpoint (in degrees):

`float railLength;`
`stepper.moveToEnd(CW,0.9);`
`stepper.encoder.setHome();`
`railLength = stepper.moveToEnd(CCW,0.9);`

the "0.9" parameter is used to set the stall detection sensitivity, and should be between 0 and 1, with lower numbers being more sensitive to stall detections.

I hope the above was helpful !

Best Regards,
Thomas Olsen

thehelpless, 2019-12-05:
Hi Thomas,

Thanks for your input / help and the time taken to read my mail! - the code example you provided worked perfectly, 

`#include <uStepperS.h>

uStepperS stepper;
float railLength;
int railSteps;

void setup() {
  stepper.setup();
  Serial.begin(9600);
  
   while(!Serial.available());
  stepper.moveToEnd(CCW,0.9);
  stepper.encoder.setHome();
  railLength = stepper.moveToEnd(CW,0.9);
  Serial.println(railLength);
  railSteps = railLength/1.8;

}`

Thanks again

closed 2019-12-05 by thehelpless