uStepper/uStepperS #19: StallGuard2 support?

greyltc, 2020-02-01:
It seems like your stall detection implementation here could be improved. Why not make use of the StallGuard2 feature in the TMC5130A? That should be much more robust and performant than what you have now.

uStepper, 2020-02-04:
Hi,

We did it the way we did to make the implementation the same for both the S and the S-lite. But we are looking in to stallguard already as we could use the processing power for other tasks than stall detection.

greyltc, 2020-02-04:
Also, the driver IC can keep track of the encoder counts so you don't have to do that on the microcontroller.

uStepper, 2020-02-05:
Hi,

I agree with you, that could be helpful, however there are a few issues, why this may not be an ideal solution

first of all we would loose 2 bits of resolution by using the ABI interface of the encoder.

secondly, one of our current issues is noise on the encoder readings, which we are currently working on solving by adding a filter to the encoder output. this noise will be present on the ABI output aswell, and since it goes directly into the driver chip, we won't be able to mitigate the problem at all

closed 2021-01-22 by uStepper