uStepper/uStepperS #22: Difference between MovedAngle and motor steps

Nestilaos, 2020-05-28:
Hi,

I observed with release 2.0.0 a difference between the moved angle and the steps ran by the motor. I am quite sure that the motor did not loose steps.

When running 51200 steps into one direction and then going back the initial angle is missed by approx. 20-40 deg according to getAngleMoved().

Is this anything known?

Regards.

Nestilaos, 2020-05-29:
I found the issue. The distance between encoder and magnet was too high. The status bit for "weak magnet field" was set.

Neither the "old" mounting bracket from uStepper/uStepper S-lite nor the newer one that allows permanent mounting can be used without modifications. The distance between magnet and encoder is just too high.

I could put some stripes of Tesa below the magnet. Now everything is working correctly as far as I can tell.

uStepper, 2020-07-11:
Hi,

Yeah we have noticed that the magnet strength is on the weak side. We cant recall this with our stock motors here though...

closed 2020-07-11 by uStepper

sliangder, 2021-05-09:
> I found the issue. The distance between encoder and magnet was too high. The status bit for "weak magnet field" was set.
>
> Neither the "old" mounting bracket from uStepper/uStepper S-lite nor the [...]