uStepper/uStepperS #26: Error compiling Continous.ino  / EEPROM / crtatmega328pb.o

LibrEars, 2021-01-22:
Hi uStepper-Team,

my trial to get the uStepperS working failed. I am running Debian 10 GNU/Linux and installed a new Arduino IDE via flatpack /flathub (v1.8.13) and followed the [steps](https://github.com/uStepper/uStepperS#instructions-for-users-new-to-ustepper) to install the uStepperS hard- and software.

Then I downloaded the Continous.ino example which does not compile (see error output below) - I even flashed the bootloader to be sure.

Do you now what might cause the error?  Also I am confused with the description of the EEPROM usage - an example-code or link for further information would help.

Thank you very much
LibrEars

```
Arduino: 1.8.13 (Linux), Board: "uStepperS"

/app/Arduino/arduino-builder -dump-prefs -logger=machine -hardware /app/Arduino/hardware -hardware /home/user/.arduino15/packages -tools /app/Arduino/tools-builder -tools /app/Arduino/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /app/Arduino/libraries -libraries /home/user/Arduino/libraries -fqbn=uStepper:avr:uStepperS -vid-pid=10C4_EA60 -ide-version=10813 -build-path /tmp/arduino_build_119629 -warnings=all -build-cache /tmp/arduino_cache_755599 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/app/Arduino/hardware/tools/avr -verbose /home/user/Examples/Continous.ino/Continous/Continous.ino
/app/Arduino/arduino-builder -compile -logger=machine -hardware /app/Arduino/hardware -hardware /home/user/.arduino15/packages -tools /app/Arduino/tools-builder -tools /app/Arduino/hardware/tools/avr -tools /home/user/.arduino15/packages -built-in-libraries /app/Arduino/libraries -libraries /home/user/Arduino/libraries -fqbn=uStepper:avr:uStepperS -vid-pid=10C4_EA60 -ide-version=10813 -build-path /tmp/arduino_build_119629 -warnings=all -build-cache /tmp/arduino_cache_755599 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino7.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/app/Arduino/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/app/Arduino/hardware/tools/avr -verbose /home/user/Examples/Continous.ino/Continous/Continous.ino
Using board 'uStepperS' from platform in folder: /home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1
Using core 'arduino' from platform in folder: /app/Arduino/hardware/arduino/avr
Detecting libraries used...
/app/Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_USTEPPER_S -DARDUINO_ARCH_AVR -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/tools/include -D__AVR_DEV_LIB_NAME__=m328pb -I/app/Arduino/hardware/arduino/avr/cores/arduino -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/variants/ustepperS /tmp/arduino_build_119629/sketch/Continous.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for uStepperS.h: [uStepper_S@2.2.0]
ResolveLibrary(uStepperS.h)
  -> candidates: [uStepper_S@2.2.0]
/app/Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_USTEPPER_S -DARDUINO_ARCH_AVR -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/tools/include -D__AVR_DEV_LIB_NAME__=m328pb -I/app/Arduino/hardware/arduino/avr/cores/arduino -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/variants/ustepperS -I/home/user/Arduino/libraries/uStepper_S/src /tmp/arduino_build_119629/sketch/Continous.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for EEPROM.h: [EEPROM@2.0 EEPROM@2.0]
ResolveLibrary(EEPROM.h)
  -> candidates: [EEPROM@2.0 EEPROM@2.0]
/app/Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_USTEPPER_S -DARDUINO_ARCH_AVR -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/tools/include -D__AVR_DEV_LIB_NAME__=m328pb -I/app/Arduino/hardware/arduino/avr/cores/arduino -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/variants/ustepperS -I/home/user/Arduino/libraries/uStepper_S/src -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/libraries/EEPROM/src /tmp/arduino_build_119629/sketch/Continous.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Using cached library dependencies for file: /home/user/Arduino/libraries/uStepper_S/src/uStepperDriver.cpp
Using cached library dependencies for file: /home/user/Arduino/libraries/uStepper_S/src/uStepperEncoder.cpp
Using cached library dependencies for file: /home/user/Arduino/libraries/uStepper_S/src/uStepperS.cpp
Using cached library dependencies for file: /home/user/Arduino/libraries/uStepper_S/src/uStepperServo.cpp
Generating function prototypes...
/app/Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -flto -w -x c++ -E -CC -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_USTEPPER_S -DARDUINO_ARCH_AVR -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/tools/include -D__AVR_DEV_LIB_NAME__=m328pb -I/app/Arduino/hardware/arduino/avr/cores/arduino -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/variants/ustepperS -I/home/user/Arduino/libraries/uStepper_S/src -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/libraries/EEPROM/src /tmp/arduino_build_119629/sketch/Continous.ino.cpp -o /tmp/arduino_build_119629/preproc/ctags_target_for_gcc_minus_e.cpp -DARDUINO_LIB_DISCOVERY_PHASE
/app/Arduino/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_119629/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/app/Arduino/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328pb -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_USTEPPER_S -DARDUINO_ARCH_AVR -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/tools/include -D__AVR_DEV_LIB_NAME__=m328pb -I/app/Arduino/hardware/arduino/avr/cores/arduino -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/variants/ustepperS -I/home/user/Arduino/libraries/uStepper_S/src -I/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/libraries/EEPROM/src /tmp/arduino_build_119629/sketch/Continous.ino.cpp -o /tmp/arduino_build_119629/sketch/Continous.ino.cpp.o
cc1plus: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
cc1plus: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
In file included from /home/user/Arduino/libraries/uStepper_S/src/uStepperS.h:152:0,
                 from /home/user/Examples/Continous.ino/Continous/Continous.ino:9:
/home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/libraries/EEPROM/src/EEPROM.h:145:20: warning: 'EEPROM' defined but not used [-Wunused-variable]
 static EEPROMClass EEPROM;
                    ^~~~~~
Compiling libraries...
Compiling library "uStepper_S"
Using previously compiled file: /tmp/arduino_build_119629/libraries/uStepper_S/uStepperDriver.cpp.o
Using previously compiled file: /tmp/arduino_build_119629/libraries/uStepper_S/uStepperEncoder.cpp.o
Using previously compiled file: /tmp/arduino_build_119629/libraries/uStepper_S/uStepperS.cpp.o
Using previously compiled file: /tmp/arduino_build_119629/libraries/uStepper_S/uStepperServo.cpp.o
Compiling library "EEPROM"
Compiling core...
[... cut ...]
/app/Arduino/hardware/tools/avr/bin/avr-objcopy:/app/Arduino/hardware/tools/avr/avr/lib/avr5/crtatmega328pb.o: Das Dateisystem ist nur lesbar
Multiple libraries were found for "EEPROM.h"
 Used: /home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/libraries/EEPROM
 Not used: /app/Arduino/hardware/arduino/avr/libraries/EEPROM
Using library uStepper_S at version 2.2.0 in folder: /home/user/Arduino/libraries/uStepper_S 
Using library EEPROM at version 2.0 in folder: /home/user/.arduino15/packages/uStepper/hardware/avr/2.2.1/libraries/EEPROM 
exit status 1
Error compiling for board uStepperS.

```

uStepper, 2021-01-22:
Hi LibrEars,

The problem is not with the bootloader or the uStepper hardware/software.
The problem is that the following file is not readable on your arduino IDE installation:
   /app/Arduino/hardware/tools/avr/avr/lib/avr5/crtatmega328pb.o

I would suggest you to install the portable arduino IDE, downloaded directly from the arduino.cc website instead of the package manager you used.

Best Regards,
Thomas Olsen

LibrEars, 2021-05-18:
This solved the problem - Thank you!

Hint: You can use a virtual machine if you do want to sandbox the IDE

closed 2021-05-18 by LibrEars