uStepper/uStepperS #31: Can't get interrupts to work

LibrEars, 2021-05-18:
Hi all,

i am trying to attach a [mechanical kill-switch](https://de.rs-online.com/web/p/mikroschalter/6821529/) (pulldown) to my setup in order to stop the motor in case my 'platform crashes into an object'.

The code works with an arduino but sadly not with the uStepperS. The digital pin itself works so I guess there is a problem with the interrupts..

Can somebody help :)?

Here is the code:

```
// Code used to test a kill-switch (pulldown configuration - with an external 10k resistor) with uStepperS
 

//include libraries
#include <math.h>

//Stop-switches definitions
#define stopSwPin 3 // pin with interrupt

//global variables
volatile int stopMove=0;        //stop-switch output
void stop_isr();
void waitForSerial();

//runs once
void setup() {
  Serial.begin(115200); //115200 is baud rate of serial comm

  //setup stop switches
  pinMode(stopSwPin, INPUT);
  attachInterrupt(digitalPinToInterrupt(stopSwPin),stop_isr,RISING);
  stopMove=0;
}

//runs forever
void loop() {

  int i,input_char;

  Serial.println("\nStart test switches? [y/n]");
  waitForSerial();
  
  for(i=0; i<5; i++){

    //check if stop switch was pressed
    if(stopMove==HIGH){
      Serial.println("MOTOR STOPPED");
      break;
    }
    
    }

    Serial.println("\nContinue to next test? [y/n]");
    waitForSerial();

}

//functions

void waitForSerial(){
  char input_char;
  
  while(Serial.available()==0);
  do{
    input_char= (char)Serial.read();
  }
  while(input_char!='y');
}

//stop switch interrupt handler
void stop_isr(){ //keep this function small!
 
  stopMove=HIGH;

}
```

LibrEars, 2021-05-21:
hi again,

i checked with another code for a simple button and LED toggle ([from here](https://www.electronicshub.org/arduino-interrupts-tutorial/)).  Still it works on the arduino (here I used pin 13 for the LED) but not for the uStepperS :/

```
int ledPin = A2;
int buttonPin = 3; // use simple button (rotary)

int ledToggle;
int previousState = HIGH;
unsigned int previousPress;
volatile int buttonFlag;
int buttonDebounce = 20;

void setup() 
{
  //interrupts();
  pinMode(ledPin, OUTPUT);
  pinMode(buttonPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(3), button_ISR, CHANGE);
}

void loop() 
{
  if((millis() - previousPress) > buttonDebounce && buttonFlag)
  {
    previousPress = millis();
    if(digitalRead(buttonPin) == LOW && previousState == HIGH)
    {
      ledToggle =! ledToggle;
      digitalWrite(ledPin, ledToggle);
      previousState = LOW;
    }
    
    else if(digitalRead(buttonPin) == HIGH && previousState == LOW)
    {
      previousState = HIGH;
    }
    buttonFlag = 0;
  }
}

void button_ISR()
{
  buttonFlag = 1;
}
```

LibrEars, 2021-06-11:
Hi Again,

i got some progress by using [Pin Change Interrupts](https://www.teachmemicro.com/arduino-interrupt-tutorial/). They are different than normal Interrupts ("might be a tiny bit slower and not that reliable because of that detection overhead (talking about micro seconds)") but there is an [arduino library](https://www.arduino.cc/reference/en/libraries/pinchangeinterrupt/).

Still it is very confusing and frustrating that normal interrupts do not work. Could it be, that they are already in use by the [dropin feature](https://ustepper.com/docs/usteppers/html/classuStepperS.html#a7d9af270acb55589631319b0f5460276)? In my understanding they should not as long as you do not use this feature actively in your code... ?

Here is an example code for Pin Change Interrupts:

```
#include "PinChangeInterrupt.h"

// Choose a valid PinChangeInterrupt pin of your Arduino board
#define buttonPin 3
int LEDPin = A2; 

void setup() 
{

  // set pin to input with a pullup, led to output
  pinMode(buttonPin, INPUT_PULLUP);
  pinMode(LEDPin, OUTPUT);
  
  // Manually blink once to test if LED is functional
  blinkLed();
  delay(1000);
  blinkLed();

  // Attach the new PinChangeInterrupt and enable event function below
  
  
}

void loop() 
{
 // Nothing to do here
}

// ISR
void blinkLed(void) {
  // Switch Led state
  digitalWrite(LEDPin, !digitalRead(LEDPin));
}
```

uStepper, 2023-10-07:
Hi,

The problem is in the "digitalPinToInterrupt(3)" function, having an implementation error.

this is fixed in the development branch of the hardware package, which will be released soon

Best Regards,
uStepper Team

closed 2023-10-07 by uStepper