uStepper/uStepperS #32: i2C example - Grove-LCD display (RGB Backlight)?

LibrEars, 2021-05-18:
Hi all,

did someone get the I2C ports SDA1 and SCL1 (Digital pins D7 & D8) of the uStepperS up running? I tried to connect a grove-LCD display (RGB Backlight) but it did not work so far.

The examples of the [grove library](https://github.com/Seeed-Studio/Grove_LCD_RGB_Backlight) did not work and I also tried without success to detect the I2C ports in general with the i2c_scanner from the arduino uno examples (with and without the pins connected, code see below)

The scanner uses the wire library. Does someone knows if this should work in principle or is there more hard stuff necessary to get it running? 

```
// --------------------------------------
// i2c_scanner
//
// Version 1
//    This program (or code that looks like it)
//    can be found in many places.
//    For example on the Arduino.cc forum.
//    The original author is not know.
// Version 2, Juni 2012, Using Arduino 1.0.1
//     Adapted to be as simple as possible by Arduino.cc user Krodal
// Version 3, Feb 26  2013
//    V3 by louarnold
// Version 4, March 3, 2013, Using Arduino 1.0.3
//    by Arduino.cc user Krodal.
//    Changes by louarnold removed.
//    Scanning addresses changed from 0...127 to 1...119,
//    according to the i2c scanner by Nick Gammon
//    https://www.gammon.com.au/forum/?id=10896
// Version 5, March 28, 2013
//    As version 4, but address scans now to 127.
//    A sensor seems to use address 120.
// Version 6, November 27, 2015.
//    Added waiting for the Leonardo serial communication.
//
//
// This sketch tests the standard 7-bit addresses
// Devices with higher bit address might not be seen properly.
//

#include <Wire.h>

void setup() {
  Wire.begin();

  Serial.begin(9600);
  while (!Serial); // Leonardo: wait for serial monitor
  Serial.println("\nI2C Scanner");
}

void loop() {
  int nDevices = 0;

  Serial.println("Scanning...");

  for (byte address = 1; address < 127; ++address) {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    byte error = Wire.endTransmission();

    if (error == 0) {
      Serial.print("I2C device found at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.print(address, HEX);
      Serial.println("  !");

      ++nDevices;
    } else if (error == 4) {
      Serial.print("Unknown error at address 0x");
      if (address < 16) {
        Serial.print("0");
      }
      Serial.println(address, HEX);
    }
  }
  if (nDevices == 0) {
    Serial.println("No I2C devices found\n");
  } else {
    Serial.println("done\n");
  }
  delay(5000); // Wait 5 seconds for next scan
}
```

LibrEars, 2021-06-25:
I made some progress. Turns out I should have used SDA0 and SDA1 from the start and then it works.

Also I found out, that the analog pins A0 and A2 are not labeled right on the board! (they need to be swaped)! I will open a new issue for that.

Also pin D7 (SCL1) and D8 (SDA1) are somehow always HIGH (missing pullup resistor?) which causes problems in my setup (not enough pins). If you for some reason need to use the SCL1 and SDA1 pins for the display (or similar) as well, I found a way :)

This is not possible by default due to missing firmware. Therefore we need to add support for the microcontroller. Here is what I did:

1. Go to the folder \Arduino\libraries and copy the folder \Grove_-_LCD_RGB_Backlight and rename it to \Grove_-_LCD_RGB_Backlight_Serial1
2. In the Folder \Grove_-_LCD_RGB_Backlight_Serial1 rename the files rgb_lcd.cpp and rgb_lcd.h to rgb_lcd1.cpp and rgb_lcd1.h
3. Open the file rgb_lcd1.cpp with an text editor and change the lines  #include <Wire.h> to #include <Wire1.h> #include "rgb_lcd.h" to #include "rgb_lcd1.h"
Also add the line
extern TwoWire1 Wire1; // insert this line is the modification to include Wire1 (SDA1 / SCL1) .…
    5. Then press ‘Ctrl+h’  to replace all function calls starting with Wire.xxx() with Wire1.xxx()
! Make sure to not replace the lines added above a second time
    6. From now on in the arduino code always include #include <Wire1.h> and #include "rgb_lcd1.h"

Sources:
    • https://github.com/MCUdude/MiniCore This lib adds support for the  ATmega328PB, in detail for serial communication with the lib Wire1.h
    • https://www.instructables.com/How-to-share-i2C-connection-on-Due/ To get the display running, we need to change its lib rgb_lcd.h

LibrEars, 2021-06-25:
https://github.com/uStepper/uStepperS/issues/35

uStepper, 2023-08-13:
We will correct the silk on any future revisions of S. The high level on D7 and D8 is caused by pull-up resistors for the i2c.

closed 2023-08-13 by uStepper