uStepper/uStepperS #34: Motor holding will sometimes turn off after stepping

BeebBenjamin, 2021-06-22:
Sometimes after stepping forward a few times, the hold will click off then come back on and get into a state where it toggles until the usteper is restarted.  Do you know what might be causing this? Here is my code:

```
#include <uStepperS.h>
  
  uStepperS stepper;
  
  int angle = 10;
  int angles[] = {360, 180, 120, 90, 72, 60, 45, 40, 36, 30, 24, 20, 18, 15, 12, 10, 9, 8, 6, 5, 4, 3, 2, 1};
  int cmd;
  int angle_index = 15;
  
  void setup() {
    stepper.setup(NORMAL,200,0,0,0,16,false, 0,80,80);
    stepper.setMaxAcceleration(5);
    stepper.setMaxVelocity(250);
    stepper.setBrakeMode(FREEWHEELBRAKE);
    Serial.begin(9600);
  }
  
  void loop() {
    if (Serial.available() > 0) {
      cmd = Serial.read() - '0';
      
      switch(cmd){
        case 0:
          stepper.setBrakeMode(HARDBRAKE, 80);
          while(stepper.getMotorState()== 1);
          Serial.write(6);
          break;
        case 1:
          stepper.moveAngle(angle);
          stepper.setBrakeMode(HARDBRAKE, 80);
          while(stepper.getMotorState()== 1);
          Serial.write(6);
          break;
        case 2:
          stepper.moveAngle(-angle);
          stepper.setBrakeMode(HARDBRAKE, 80);
          while(stepper.getMotorState()==1);
          Serial.write(6);
          break; 
        case 3:
          stepper.stop();
          stepper.setBrakeMode(FREEWHEELBRAKE);
          break;
        case 4:
          if (angle_index == 23){
            angle_index = 0;
          }
          if (angle_index < 23){
            angle_index = angle_index + 1;
          }
          angle = angles[angle_index];
          break;
        case 5:
          if (angle_index == 0){
            angle_index = 23;
          }
          if (angle_index >  0){
            angle_index = angle_index - 1;
          }
          angle = angles[angle_index];
          break;
      }
    }
  }

BeebBenjamin, 2021-06-23:
Tested on the following code changes:

```
#include <uStepperS.h>

uStepperS stepper;

int angle = 10;
int angles[] = {360, 180, 120, 90, 72, 60, 45, 40, 36, 30, 24, 20, 18, 15, 12, 10, 9, 8, 6, 5, 4, 3, 2, 1};
int cmd;
int angle_index = 15;

void setup() {
  stepper.setup(NORMAL,200,0,0,0,16,false, 0,80,80);
  stepper.setMaxAcceleration(5);
  stepper.setMaxVelocity(250);
  stepper.setBrakeMode(FREEWHEELBRAKE);
  Serial.begin(9600);
}

void loop() {
  if (Serial.available() > 0) {
    cmd = Serial.read() - '0';
    
    switch(cmd){
      case 0:
        stepper.setBrakeMode(HARDBRAKE, 80);
        Serial.write(6);
        break;
      case 1:
        stepper.moveAngle(angle);
        while(stepper.getMotorState()== 1);
        stepper.setBrakeMode(HARDBRAKE, 80);
        Serial.write(6);
        break;
      case 2:
        stepper.moveAngle(-angle);
        while(stepper.getMotorState()==1);
        stepper.setBrakeMode(HARDBRAKE, 80);
        Serial.write(6);
        break; 
      case 3:
        stepper.stop();
        stepper.setBrakeMode(FREEWHEELBRAKE);
        Serial.write(6);
        break;
      case 4:
        if (angle_index == 23){
          angle_index = 0;
        }
        if (angle_index < 23){
          angle_index = angle_index + 1;
        }
        angle = angles[angle_index];
        Serial.write(6);
        break;
      case 5:
        if (angle_index == 0){
          angle_index = 23;
        }
        if (angle_index >  0){
          angle_index = angle_index - 1;
        }
        angle = angles[angle_index];
        Serial.write(6);
        break;
    }
  }
}

The ustepper goes into an unresponsive state after sitting idle, and cycles between the motor engaged or off.  It happens when I run the serial link through the Arduino IDE and when I am using Python Serial.

BeebBenjamin, 2021-06-23:
Update: I can stop the cycle by sending a 1 or a 2 to turn the motor, but it will once again go into cycling on and off after sitting idle with the break on for a short while.

BeebBenjamin, 2021-06-24:
I do wonder if I am seeing this manifest: https://github.com/uStepper/uStepperS/issues/7 so I have lowered the current to 50% and am doing some tests.

uStepper, 2023-08-13:
As you indicate on the last post, active cooling or lower current settings is required for "long" time braking.

closed 2023-08-13 by uStepper