uStepper/uStepperS #46: Add board footprint

kriskeillor, 2023-02-06:
Hello,
I am developing a shield for the uStepper S that adds bluetooth functionality. It would be very helpful to have a complete, dimensioned footprint of the uStepper S and uStepper S-Lite (if there's any difference) to properly align the headers on my shield.

As it stands, the only information I can find is the overall size and hole placement (41.8mm and 36mm, respectively). 

Information about the three headers is necessary. For folks integrating enclosures, the button and power/USB port placements would also be helpful.

Thank you!

uStepper, 2023-02-08:
Hi Kris,

I have attached a step file of the uStepper S pcb here. the connector and hole placement are identical for the S-lite
[uStepper.zip](https://github.com/uStepper/uStepperS/files/10689869/uStepper.zip)

I hope this helps !

Best Regards,
Thomas Olsen

uStepper, 2023-02-13:
Hi,A step model with precise dimensions is available here:uStepperS/Hardware at master · uStepper/uStepperSgithub.comBest regards, Mogens

kriskeillor, 2023-02-19:
Hi Thomas,

Thank you for sharing that! 

I was aware of the step file. I don't currently have access to, or skill with, a 3d modeling program that can quickly tell me the dimensions. A board outline meant for use with typical electrical schematic capture programs (Altium and KiCad would be a great start) would be a blessing for electrical engineers with less mechanical aptitude. 

Fortunately, I got pretty accurate measurements using a digital caliper.

Again thank you for the helpful reply!
Best,
Kris

closed 2023-02-19 by kriskeillor